frc5885/2025-Poseidon #12: Algae Gripper

johnthibert, 2025-01-06:


marcotron64, 2025-01-14:
January 13:

Done:
- Basic shape
- Motor mounts
- Shaft/Bearing mounts

To do:
- Add another panel to increase width
- Add slide slot
- Add struts/mounting holes
- Cut speed holes

[image]

marcotron64, 2025-01-15:
January 14

Done: 
- Added third plate
- Added more prefabs to assembly (wheels and pulleys and what have you)
- Added slide slot

To do:
- Add struts/mounting holes
- Possibly change slide slot to curved linear guide rail
- Cut speed holes

[image]

johnthibert, 2025-01-16:
Link to ISO Tables: https://simplybearings.co.uk/shop/Info-Pages-ISO-Limits/c4746_4779/index.html

marcotron64, 2025-01-16:
January 15:

Done:
- Decided on a curved linear rail
- Created rail from profile from a rail found online

To do:
- Finalize rail (mounting holes + whatever else)
- Create bearing wheel(?)

closed 2025-02-10 by h0im3die