frc5885/2025-Poseidon #14: Finish Design of 2025 Chassis

johnthibert, 2025-01-06:
- [ ] Top and Bottom panels for electronic boards.

zjs0221, 2025-01-07:
Needs holes:
- Robo RIO (w/ NAV-X)
- Battery
- PDP (REV)
- VRM
- SPARK MAX (x8)

Needs space:
- Breaker (can be on battery)
- Ethernet switch
- Raspberry Pi box + battery

May be mounted elsewhere:
- Radio
- Safety light

johnthibert, 2025-01-07:
I love these ideas @zjs0221, you're killing it! That ethernet switch would work great, ill buy it for the team and you can design up an ethernet cable holder

zjs0221, 2025-01-14:
First iteration of case, to be changed to a screw system for security.
[image]

Ideas:

[image]
multi part system where a bracket is attached to each cord and screwed into case

[image]
probably not stable enough, simple though.

John's idea (no image) I'm not entirely sure how to design the clip to clamp onto the cord. 
- maybe clamp where the bending grooves are?
- or just make them 3 cm out form the case.

johnthibert, 2025-01-15:
@zjs0221 I ordered this switch, I know the dimensions probably changed, but this one is much more power efficient and runs on 5V, which is the same voltage as the Orange Pis

zjs0221, 2025-02-06:
Here's the case before the pattern.

[image]

zjs0221, 2025-02-06:
Case features:
- team number vents
- labelled power cord hole
- opening for indicator lights
- mounting holes (10-32s)
- holes for threaded inserts (M4s)

closed 2025-02-06 by zjs0221