frc5885/2025-Poseidon #15: Bumper Design and Concepting

johnthibert, 2025-01-06:
Assigned to Sophia Russo, Celeste, and Ryan?

CelesteDee07, 2025-01-08:
Tuesday, Jan 7th
Today we talked to Bruce about some design ideas we had for the bumpers. We tweaked the design a bit and started cading the backing of our bumpers. Then we downloaded cad for the clips we had found to try out this year. We downloaded the file so Bruce could get the wood backing cut out for Thursday. We also read over the rules a ton more also. We will keep looking it of to see if we can add extra materials other then wood to the backing.

CelesteDee07, 2025-01-08:
Monday, Jan 6th
Today is day 1 of design and planning. Me, Vic and Ryan read over the rules for bumpers (writing down all important dimensions and restrictions)and then started researching the best way to have easy to remove bumpers that are firm enough to give our robot some extra defense. We found very effective and cheap Clips that we want to try out. We also went over the way we wanted to pad the bumpers (using floor files and noodles), we made a drawing of our concept and wrote down the clip info.

CelesteDee07, 2025-01-08:
[image]
[image]
[image]
[image]

DuckyNumber1, 2025-01-10:
Thursday, Jan 9th
Today Bruce brought the wood pieces for the bumper that we used CAD to create. We glued them together and made sure that they we're square. We are letting the glue dry so that they are ready for us to put fiberglass on them next time.

CelesteDee07, 2025-01-11:
Saturday, Jan 11th

Today we worked of the frame for the bumpers! We took the wood frame we glued together on Thursday and added fiber glass to it. Bruce tought us how to properly saturate the strips in a resin we mixed and add it to the wood. We covered two sides on each bumper and clamped it to remove excess resin. We also added some to the inside joints of the frame since we had extra resin. Overall it was a pretty solid day!

DuckyNumber1, 2025-01-14:
https://www.mcmaster.com/1872A113/
Name: Slide Snap Draw Latch
It is for latching the bumpers down to the frame
Dimensions: length: 1 3/4", width: 11/16"
Cost: 10.30$
Part number: 1872A113

KillaKuli, 2025-01-14:
Hi, you can buy the part # for much cheaper on Pegasus Auto Racing Supplies
website:
https://www.pegasusautoracing.com/productdetails.asp?RecID=4660&srsltid=AfmBOoryM3Z_-6QxnB4AgWsnJ-g5upkSiWxJO1KPpzUo8PKdFZ3bJlyM

closed 2025-02-09 by johnthibert