frc5885/2025-Poseidon #17: Robot Battery Cart Design

johnthibert, 2025-01-06:


zjs0221, 2025-01-07:
Here are previous designs for the battery cart, they no longer fit the requirements, but feel free to use any elements or concepts!

- [https://cad.onshape.com/documents/0f5faa7b37d7469f46de3151/w/500beaf497b75716d460ef9d/e/a9571d395f222b97cdeca518?renderMode=0&uiState=677c721c248f3576864ba013](url)
- [https://cad.onshape.com/documents/7faff71dcb1d937b7360696e/w/368685d044e1de5be26cdfbe/e/bf60a34ddfe0768b6616e16c?renderMode=0&uiState=677c72848ed62e6f7bcb8449](url)

*They are both on the old Wiredcats folder on Onshape.

jpeters5, 2025-01-07:
**Robot Battery Cart**

### Objectives

1.	Mobile structure to hold, transport, and charge the batteries needed for the FRC robot in a competition.

### Constraints

1.  “Uline Loop Handle Steel Hand Truck” as base platform
      - Model #: H-1786
      - Load Capacity: 800lbs
      - Height: 51”
      - Weight: 44lbs
      - Wheels: 10" Rubber (400 lb cap): H-3358

### Requirements

1.  Needs to hold at least 10 batteries
2.  Has 1 main power bar that plugs into a wall electrical socket
3.  Travel storage on its back
4.  No wider than 31” so it can go through a standard door opening with 1” clearance
5.  Must be stable standing up.
     - Low center of gravity while loaded
     - Components located as close as possible to lateral center line and back frame.

closed 2025-01-13 by johnthibert